gnuradio/gr-recipes #13: GNURadio fails on missing thrift dependency

1am, 2016-04-07:
Hi

I've been trying to install gnuradio on Ubuntu via PyBOMBS but it fails on some thrift dependency with the following error:

```
Linking CXX executable gr_runtime_test
libgnuradio-runtime-3.7.10git.so.0.0.0: undefined reference to `apache::thrift::async::TConcurrentClientSyncInfo::getPending(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, apache::thrift::protocol::TMessageType&, int&)'
libgnuradio-runtime-3.7.10git.so.0.0.0: undefined reference to `apache::thrift::async::TConcurrentClientSyncInfo::updatePending(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, apache::thrift::protocol::TMessageType, int)'
collect2: error: ld returned 1 exit status
gnuradio-runtime/lib/CMakeFiles/gr_runtime_test.dir/build.make:98: recipe for target 'gnuradio-runtime/lib/gr_runtime_test' failed
make[2]: *** [gnuradio-runtime/lib/gr_runtime_test] Error 1
CMakeFiles/Makefile2:597: recipe for target 'gnuradio-runtime/lib/CMakeFiles/gr_runtime_test.dir/all' failed
make[1]: *** [gnuradio-runtime/lib/CMakeFiles/gr_runtime_test.dir/all] Error 2
Makefile:146: recipe for target 'all' failed
make: *** [all] Error 2
PyBombs.monitor_process() - DEBUG - Thread signaled termination or returned
PyBombs.monitor_process() - DEBUG - Return value: 2
PyBombs.Packager.source - ERROR - Build failed. See output above for error messages.
PyBombs.Packager.source - ERROR - Problem occurred while building package gnuradio:
Build failed.
PyBombs.install - ERROR - Error installing package gnuradio. Aborting.
```

The thing is that running `pybombs -p default -vvv install gnuradio` pulls apache-thrift as a dependency so I'm not sure what am I missing?

ckuethe, 2016-04-08:
"libgnuradio-runtime-3.7.10git.so.0.0.0: undefined reference to `apache::thrift::async::TConcurrentClientSyncInfo::getPending(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, apache::thrift::protocol::TMessageType&, int&)'"

Smells like a missing -lthrift, though I'm not sure why. Can you try get the actual gcc command that failed?

1am, 2016-04-10:
I've tried again today on a fresh Ubuntu Mate 15.10 x64 and it worked fine. Probably something was messed up with my previous Ubuntu installation. Good thing that I could get rid of it and start from a fresh distro.

closed 2016-04-10 by 1am

courtarro, 2017-05-15:
I was experiencing this issue on a machine that was upgraded from Ubuntu 14.04 (trusty) to 16.04 (xenial) and whose GR installation had gotten quite stale.

**Workaround**:

1. Update Apache Thrift manually

        pybombs update apache-thrift

2. Reload my prefix

        source ~/prefix/setup_env.py

3. Then continue building `gnuradio`

        cd ~/prefix/src/gnuradio/build
        make -j4