gnuradio/gr-recipes #45: gmp.lwr:  404 Not Found

scateu, 2016-07-28:
```
wget http://archive.ubuntu.com/ubuntu/pool/main/g/gmp/gmp_4.3.2+dfsg.orig.tar.gz
--2016-07-28 10:05:53--  http://archive.ubuntu.com/ubuntu/pool/main/g/gmp/gmp_4.3.2+dfsg.orig.tar.gz
Resolving archive.ubuntu.com (archive.ubuntu.com)... 91.189.88.162, 91.189.88.161, 91.189.88.152, ...
Connecting to archive.ubuntu.com (archive.ubuntu.com)|91.189.88.162|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-07-28 10:05:54 ERROR 404: Not Found.
```

https://github.com/gnuradio/gr-recipes/blob/master/gmp.lwr#L26

ckuethe, 2016-08-17:
could you test a different gmp URL, for example `https://ftp.gnu.org/gnu/gmp/gmp-6.1.0.tar.bz2` and let us know if you get a good build?

mbr0wn, 2016-09-06:
I updated the URL and am crossing my fingers.

closed 2016-09-06 by mbr0wn

scateu, 2017-01-09:
It works! Thank you.