gnuradio/gr-recipes #59: Missing dependencies on CentOS 7

Flamewires, 2016-09-17:
<pre>
user@host$ sudo docker run --net=host -it centos /bin/bash
[root@host /]# yum install epel-release git
[root@host /]# yum install python-pip
[root@host /]# pip install pybombs
[root@host /]# pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git 
[root@host /]# pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-etcetera.git 
[root@host /]# pybombs prefix init /gnuradio -a myprefix -R gnuradio-default
</pre>

<pre>
PyBOMBS._process_thread() - DEBUG - Executing command `['sudo', '-H', 'yum', '-y', 'install', 'autoconf']'
Exception in thread Thread-1:
<...snip...>
OSError: [Errno 2] No such file or directory
</pre>

From the above, It looks like sudo is necessary as well. I believe sudo not being present is only going to happen in docker, or very restricted environments. However, later we failed on make, then on g++.  Neither of these are present in a minimal CentOS install (maybe not in desktop either?). What would be the best way to go about adding them as dependencies? They will be necessary for almost any source built package, so I'm not sure where to modify the recipes to include them, as they will only be necessary if source-building, however, they will at least be necessary for gnuradio in every case. 

Anyway, after installing them, things started building, but I think I saw a build issue with thrift. I'll have another look later and open a separate ticket.

Flamewires, 2016-09-17:
Looks like there is already a recipe for make and g++, so hopefully it just needs to be throw in as a dependency for whatever was failing. Will test this and submit pr if that's the case.

mbr0wn, 2016-09-18:
I would argue that hard-coding sudo as a requirement was a bug in the first place. This was addressed in a recent commit, and sudo can be un-configured for distros that don't have it.

mbr0wn, 2016-09-18:
Also, as for g++ and make, they are dependencies for all source builds. If they're not getting installed, that's a bug in PyBOMBS.

mbr0wn, 2016-09-18:
Specifically, the build-essentials.lwr is a requirement for all source builds.

mbr0wn, 2016-12-07:
I think this should be fixed -- please comment if that's not the case.

closed 2016-12-07 by mbr0wn