gnuradio/gr-recipes #65: gr-recipes/alsa.lwr

IAmThil, 2016-11-23:
URL needs updating

> PyBOMBS.Fetcher.wget - DEBUG - Downloading file: http://www.tux.org/pub/sites/ftp.alsa-project.org/lib/alsa-lib-1.0.27.2.tar.bz2
> PyBOMBS.Fetcher.wget - ERROR - HTTPConnectionPool(host='www.tux.org', port=80): Max retries exceeded with url: /pub/sites/ftp.alsa-project.org/lib/alsa-lib-1.0.27.2.tar.bz2 (Caused by NewConnectionError('<requests.packages.urllib3.connection.HTTPConnection object at 0x1040b1290>: Failed to establish a new connection: [Errno 61] Connection refused',))
> PyBOMBS.Packager.source - ERROR - Problem occurred while building package alsa:
> Unable to fetch recipe alsa
> PyBOMBS.install_manager - ERROR - Error installing package alsa. Aborting.

mbr0wn, 2016-11-23:
Do you want to update it?

IAmThil, 2016-11-24:
Sure, I've submitted a request.

closed 2016-11-24 by IAmThil