gnuradio/gr-recipes #68: libosmo-dsp LIBTOOL error

drws, 2016-12-20:
I'm using PyBOMBS v2.3.0 on CentOS 7 (x64) and gnuradio-stable (as well as gnuradio-default) is building and installing properly. However, libosmo-dsp build is failing due to the following error:

    PyBOMBS.install_manager - INFO - Installing package: libosmo-dsp
    Configuring: (100%)
    PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
    src/Makefile.am:8: error: Libtool library used but 'LIBTOOL' is undefined
    src/Makefile.am:8:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
    src/Makefile.am:8:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
    src/Makefile.am:8:   If 'LT_INIT' is in 'configure.ac', make sure
    src/Makefile.am:8:   its definition is in aclocal's search path.
    autoreconf: automake failed with exit status: 1
    PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
    PyBOMBS.Packager.source - ERROR - Problem occurred while building package libosmo-dsp:
    Configuration failed
    PyBOMBS.install_manager - ERROR - Error installing package libosmo-dsp. Aborting.

Autoconf as well as libtool are installed. Did I miss something?

mbr0wn, 2017-01-04:
Argh, I've seen this before, but can't put a finger on it. I think it might be a libtool version issue. Or maybe it can't find libtool? That would explain why LIBTOOL is undefined. How did you install libtool? Which version do you have?

estatz, 2017-01-04:
I've been down this road before, as well.  This issue arises for CentOS 7 because you'll have a newer version of autoconf installed in your pybombs prefix, but the libtool you want to use is system libtool installed under /usr.  autoconf in the pybombs prefix doesn't know to look in /usr/share/aclocal for your libtool macros 

The workaround I've been using is to make the libosmo-dsp configure do: `autoreconf -i -I /usr/share/aclocal`

I got distracted before I found a solution that was clean enough to check in to the libosmo-dsp recipe and/or pybombs as a true fix

drws, 2017-01-04:
@mbr0wn libtool (2.4.2-21.el7_2)  was installed from the official CentOS 7 repository with yum

@estatz adding the `-I` flag to the autoreconf line in `libosmo-dsp.lwr` also fixed it for me

mbr0wn, 2017-01-04:
Is there a way we can fix this generically?