gnuradio/gr-recipes #77: Incorrect deb package for x11 recipe

remicles, 2017-04-20:
When trying to satisfy x11 dependency (required by gr-fosphor install), I get the following error message:
```
$ pybombs --version
2.3.1a0
$ pybombs -v install x11
...
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
PyBOMBS.PackageManager - DEBUG - Checking if package x11 is installable...
PyBOMBS.PackageManager - DEBUG - Package x11 is not installable.
PyBOMBS.install_manager - ERROR - Package has no install method: x11
PyBOMBS - DEBUG - Unresolved install path.
```
The recipe contains:
```category: baseline
depends: null
satisfy:
  deb: libx11-dev && libxrender-dev
```
However, on my system, the correct capitalizations are "`libX11-dev`" and "`libxrender-dev`".
```
$ cat /etc/lsb-release
DISTRIB_ID=LinuxMint
DISTRIB_RELEASE=17.1
DISTRIB_CODENAME=rebecca
DISTRIB_DESCRIPTION="Linux Mint 17.1 Rebecca"

$ apt search "libx11-dev" "libxrender-dev"
i   libX11-dev                      - X11 client-side library (development heade
p   libX11-dev:i386                 - X11 client-side library (development heade
i   libxrender-dev                  - X Rendering Extension client library (deve
p   libxrender-dev:i386             - X Rendering Extension client library (deve
```
Note the differing capitalization.  `libX11-dev` seems to be the only related package with a capital X, for some reason.

remicles, 2017-04-20:
Previously there was an attempt to fix with aa4c27b3cfc86ae3fb28bf13abc7678bfed2b3b4, so perhaps this won't work on all systems.

mbr0wn, 2017-04-22:
Yeah, on my Ubuntu, it's lower-case. We can have both, though. `libx11-dev || libX11-dev`. Can you update the PR? I'll move this discussion to #78.

closed 2017-04-22 by mbr0wn

remicles, 2017-04-26:
@mbr0wn I've updated the commit.  I should have done it that way in the first place.  Thanks!