gnuradio/gr-recipes #85: pyqt5 install issue

mj355, 2017-06-28:
I'm trying to install RFNoC based on instructions [here](https://kb.ettus.com/Getting_Started_with_RFNoC_Development) and I'm running into an issue with PyQt5. Following is the error output: 
```
PyBOMBS.install_manager - INFO - Installing package: pyqt5
PyBOMBS.PackageManager - DEBUG - install(pyqt5, install_type=source, static=False)
PyBOMBS.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (pyqt5)
PyBOMBS.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (pyqt5)
PyBOMBS.PackageManager - OBNOXIOUS - pyqt5 has cached installed-status: False
PyBOMBS.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (pyqt5)
PyBOMBS.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (pyqt5)
PyBOMBS.PackageManager - DEBUG - Using packager source
PyBOMBS.Packager.source - DEBUG - State on package pyqt5 is 0
PyBOMBS.Fetcher - DEBUG - Fetching source for recipe: pyqt5
PyBOMBS.Fetcher - DEBUG - Switching to src directory: /home/(diirname)/rfnoc/src
PyBOMBS.Fetcher - OBNOXIOUS - Trying to fetch wget+https://sourceforge.net/projects/pyqt/files/PyQt5/PyQt-5.8.2/PyQt5_gpl-5.8.2.tar.gz
PyBOMBS.Fetcher - DEBUG - Getting fetcher for https://sourceforge.net/projects/pyqt/files/PyQt5/PyQt-5.8.2/PyQt5_gpl-5.8.2.tar.gz, type wget
PyBOMBS.Fetcher - DEBUG - Using fetcher <pybombs.fetchers.wget.Wget object at 0x7ff10c977810>
PyBOMBS.Fetcher.wget - DEBUG - Requiring packages on host system: ['python-requests']
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBOMBS.PackageManager - DEBUG - pip is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager apt
PyBOMBS.PackageManager - DEBUG - apt is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager yumdnf
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager port
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager brew
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager zypper
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pacman
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager portage
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pymod
PyBOMBS.PackageManager - DEBUG - pymod is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pkgconfig
PyBOMBS.PackageManager - DEBUG - pkgconfig is supported!
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager cmd
PyBOMBS.PackageManager - DEBUG - cmd is supported!
PyBOMBS.PackageManager - DEBUG - Using binary packagers: ['pip', 'apt', 'pymod', 'pkgconfig', 'cmd']
PyBOMBS.install_manager - DEBUG - Phase 1: Creating install tree and installing binary packages:
PyBOMBS.install_manager - OBNOXIOUS - Checking if package `python-requests' goes into tree...
PyBOMBS.PackageManager - OBNOXIOUS - python-requests has cached installable-status: True
PyBOMBS.PackageManager - OBNOXIOUS - python-requests has cached installed-status: True
PyBOMBS.install_manager - OBNOXIOUS - Installed, but no update requested. Does not go into tree.
PyBOMBS.install_manager - DEBUG - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.Fetcher.wget - DEBUG - Requirements met.
PyBOMBS.Fetcher.wget - DEBUG - Downloading file: https://sourceforge.net/projects/pyqt/files/PyQt5/PyQt-5.8.2/PyQt5_gpl-5.8.2.tar.gz
03745 kB / 03745 kB (100%)
PyBOMBS.Fetcher.wget - DEBUG - MD5: c3048e9d242f3e72fd393630da1d971a
PyBOMBS.extract_to - DEBUG - Unpacking PyQt5_gpl-5.8.2.tar.gz
PyBOMBS.extract_to - DEBUG - Common prefix: PyQt5_gpl-5.8.2
PyBOMBS.extract_to - DEBUG - Moving /tmp/tmpaoJse4/PyQt5_gpl-5.8.2 -> pyqt5
PyBOMBS.Fetcher - OBNOXIOUS - Success.
PyBOMBS.Inventory - OBNOXIOUS - Setting key source on package pyqt5 to wget+https://sourceforge.net/projects/pyqt/files/PyQt5/PyQt-5.8.2/PyQt5_gpl-5.8.2.tar.gz.
PyBOMBS.Inventory - DEBUG - Setting state to `fetched'
PyBOMBS.Inventory - DEBUG - Saving inventory to file /home/(dirname)/rfnoc/.pybombs/inventory.yml...
PyBOMBS.Requirer - DEBUG - Requiring packages on host system: []
PyBOMBS.Requirer - DEBUG - Requirements met.
PyBOMBS.Packager.source - DEBUG - Using build directory: /home/(dirname)/rfnoc/src/pyqt5
PyBOMBS.Packager.source - DEBUG - Configuring recipe pyqt5
PyBOMBS.Packager.source - DEBUG - Using vars - CommentedMap([('config_opt', ''), ('builddir', '/home/(dirname)/rfnoc/src/pyqt5')])
PyBOMBS.Packager.source - DEBUG - In cwd - /home/(dirname)/rfnoc/src/pyqt5
PyBOMBS._process_thread() - DEBUG - Executing command `$ python configure.py --confirm-license -b /home/(dirname)/rfnoc/bin -d /home/(dirname)/rfnoc/lib/python2.6/site-packages/ -v /home/(dirname)/rfnoc/share/sip/'
Querying qmake about your Qt installation...
Determining the details of your Qt installation...
This is the GPL version of PyQt 5.8.2 (licensed under the GNU General Public
License) for Python 2.7.6 on linux2.
Found the license file pyqt-gpl.sip.
Error: Make sure you have a working sip on your PATH or use the --sip argument
to explicitly specify a working sip.
PyBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
PyBOMBS.monitor_process() - DEBUG - Return value: 1
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
PyBOMBS.Packager.source - DEBUG - Configuring recipe pyqt5
PyBOMBS.Packager.source - DEBUG - Using vars - CommentedMap([('config_opt', ''), ('builddir', '/home/(dirname)/rfnoc/src/pyqt5')])
PyBOMBS.Packager.source - DEBUG - In cwd - /home/(dirname)/rfnoc/src/pyqt5
PyBOMBS._process_thread() - DEBUG - Executing command `$ python configure.py --confirm-license -b /home/(dirname)/rfnoc/bin -d /home/(dirname)/rfnoc/lib/python2.6/site-packages/ -v /home/(dirname)/rfnoc/share/sip/'
Querying qmake about your Qt installation...
Determining the details of your Qt installation...
This is the GPL version of PyQt 5.8.2 (licensed under the GNU General Public
License) for Python 2.7.6 on linux2.
Found the license file pyqt-gpl.sip.
Error: Make sure you have a working sip on your PATH or use the --sip argument
to explicitly specify a working sip.
PyBOMBS.monitor_process() - DEBUG - Thread signaled termination or returned
PyBOMBS.monitor_process() - DEBUG - Return value: 1
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package pyqt5:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package pyqt5. Aborting.
```
This is on a Ubuntu 14.04.5 installation using the latest gr-recipes pull (including the `pyqt5.lwr` recipe). I tried tracing the error and it seems that there is an issue with `sip`. When I try to install `sip` using `pybombs install sip`, the output says that the package is already installed, but I'm unable to find it in my prefix directory. 

Am I doing something wrong here?

mbr0wn, 2017-07-07:
We forgot to merge it. It should work now.

mbr0wn, 2017-07-07:
Please reopen if I misunderstood the issue.

closed 2017-07-07 by mbr0wn