gnuradio/gr-recipes #88: Apache Thrift 0.9.3 fails to build on Fedora 26

awalls-cx18, 2017-07-16:
Apache Thrift 0.9.3 fails to build on Fedora 26, because libcrypto from the latest openssl shipping with Fedora 26 has changed enough for it to matter to thrift 0.9.3's configure script.

Apparently thrift 0.10.0 has addressed this issue, and Fedora 26 provides thrift-0.10.0 and thrift-devel-0.10.0 as distro packages.  https://bugzilla.redhat.com/show_bug.cgi?id=1385702

The GNURadio wiki claims that 0.9.3 or higher is acceptable:
https://wiki.gnuradio.org/index.php/ControlPort

argilo, 2018-04-28:
The same problem occurs on Ubuntu 18.04. I tried updating the apache-thrift recipe to use Thrift 0.11.0, but then gnuradio fails to build.

To work around the problem, I simply removed apache-thrift from the dependency list in the gnuradio recipe.

mbr0wn, 2018-06-11:
Upgraded to 0.10.0.

closed 2018-06-11 by mbr0wn