gnuradio/gr-recipes #89: Ubuntu 16.04 PyQt5 Install Issue

ilhaan, 2017-07-20:
I was following RFNoC install instructions from [here](https://kb.ettus.com/Getting_Started_with_RFNoC_Development) and ran into an issue. After running `pybombs prefix init ~/rfnoc -a rfnoc -R rfnoc`, I received a success message telling me all packages were installed. However, PyQt5 was not installed and this prevented me from using `uhd_image_builder_gui.py` from [uhd-fpga](https://github.com/EttusResearch/fpga/tree/rfnoc-devel). 

Running `pybombs install pyqt5` would result in a "No packages to install' message. Weirdly, running `pybombs remove pyqt5` would give me a "Package pyqt5 is not installed. Aborting" message. I also found out that qt5 (PyQt5 dependency) was not installed and ran `pybombs install qt5` to fix this. 

After this, I manually installed PyQt5 in the rfnoc prefix and I can now open the above mentioned script. 

I did make a few other changes for this to go through and I will be submitting those changes in a pull request. I wasn't able to figure out why pybombs would not install qt5 and pyqt5 from the log messages, but I am re-running the installation soon and will be able to post detailed logs here. 

Note: I am using pybombs version 2.3.1 installed via pip.

ilhaan, 2017-07-21:
Just repeated the installation from scratch and PyQt5 is not installed again. Tried running `pybombs -v -v install pyqt5` and got this: 
```
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /etc/pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - Parsing config file failed (/etc/pybombs/config.yml).
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /home/ilhaan/.pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: ordereddict([('git-cache', '/home/ilhaan/.pybombs/gitcache'), ('elevate_pre_args', ['sudo', '-H']), ('makewidth', 4), ('packagers', 'pip,apt,pymod,pkgconfig,cmd'), ('default_prefix', 'rfnoc')])
PyBOMBS.ConfigManager - DEBUG - Template directory: /home/ilhaan/.local/lib/python2.7/site-packages/pybombs/templates
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/ilhaan/.pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Using environment variable PYBOMBS_PREFIX as prefix (/home/ilhaan/rfnoc)
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Choosing default prefix config dir: /home/ilhaan/rfnoc/.pybombs
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Inspecting config file: /home/ilhaan/rfnoc/.pybombs/config.yml
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix source dir is: /home/ilhaan/rfnoc/src
PyBOMBS.Inventory - DEBUG - Trying to load inventory file /home/ilhaan/rfnoc/.pybombs/inventory.yml...
PyBOMBS.ConfigManager.PrefixInfo - DEBUG - Prefix-local recipe dir is: /home/ilhaan/rfnoc/.pybombs/recipes
PyBOMBS.ConfigManager - DEBUG - Reading config info from file: /home/ilhaan/rfnoc/.pybombs/config.yml
PyBOMBS.ConfigManager - DEBUG - New config items: {}
PyBOMBS.ConfigManager - DEBUG - Full list of recipe locations: ['/home/ilhaan/rfnoc/.pybombs/recipes', '/home/ilhaan/.pybombs/recipes/gr-recipes', '/home/ilhaan/.pybombs/recipes/gr-etcetera', '/home/ilhaan/.pybombs/recipes/ettus', '/home/ilhaan/.local/lib/python2.7/site-packages/pybombs/recipes']
PyBOMBS.ConfigManager - DEBUG - Named recipe locations: {'ettus': 'git+https://github.com/EttusResearch/ettus-pybombs.git', 'gr-etcetera': 'git+https://github.com/gnuradio/gr-etcetera.git', 'gr-recipes': 'git+https://github.com/gnuradio/gr-recipes.git'}
PyBOMBS - INFO - PyBOMBS Version 2.3.1
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /home/ilhaan/rfnoc/.pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/home/ilhaan/rfnoc/.pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 0 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /home/ilhaan/.local/lib/python2.7/site-packages/pybombs/templates/prefix.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /home/ilhaan/.local/lib/python2.7/site-packages/pybombs/templates/sdk.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /home/ilhaan/.local/lib/python2.7/site-packages/pybombs/templates/setuptools.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /home/ilhaan/.local/lib/python2.7/site-packages/pybombs/templates/cmake.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /home/ilhaan/.local/lib/python2.7/site-packages/pybombs/templates/bootstrapautoconf.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /home/ilhaan/.local/lib/python2.7/site-packages/pybombs/templates/distutils.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /home/ilhaan/.local/lib/python2.7/site-packages/pybombs/templates/bjam.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /home/ilhaan/.local/lib/python2.7/site-packages/pybombs/templates/boostflags.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /home/ilhaan/.local/lib/python2.7/site-packages/pybombs/templates/empty.lwt
PyBOMBS.RecipeListManager - OBNOXIOUS - Adding template /home/ilhaan/.local/lib/python2.7/site-packages/pybombs/templates/autoconf.lwt
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /home/ilhaan/.pybombs/recipes/gr-recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/home/ilhaan/.pybombs/recipes/gr-recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 172 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /home/ilhaan/.pybombs/recipes/gr-etcetera
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/home/ilhaan/.pybombs/recipes/gr-etcetera' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 89 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /home/ilhaan/.pybombs/recipes/ettus
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/home/ilhaan/.pybombs/recipes/ettus' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 9 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.RecipeListManager - DEBUG - Adding recipe location: /home/ilhaan/.local/lib/python2.7/site-packages/pybombs/recipes
PyBOMBS.RecipeListManager - DEBUG - Scanning directory '/home/ilhaan/.local/lib/python2.7/site-packages/pybombs/recipes' for recipes...
PyBOMBS.RecipeListManager - DEBUG - Found 11 new recipes.
PyBOMBS.RecipeListManager - DEBUG - Loading templates.
PyBOMBS.install - DEBUG - Initializing command class for command install
PyBOMBS.PackageManager - DEBUG - Attempting to add binary package manager pip
PyBOMBS.PackageManager - DEBUG - pip is supported!
[... cut ...]
PyBOMBS.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (pyqt5)
PyBOMBS.get_recipe - OBNOXIOUS - Woohoo, this one's already cached (pyqt5)
PyBOMBS.PackageManager - DEBUG - Package pyqt5 is not installed.
PyBOMBS.remove - ERROR - Package pyqt5 is not installed. Aborting.
```
qt5 is not installed as well and I'm unable to install it using pybombs like I mentioned in my previous post. The `install` command tells me that the package is installed already and the `remove` command says it is not, just like pyqt5.

rodrigofreundt, 2018-02-16:
Would you share please how did you managed to manually install PyQt5 in the rfnoc prefix?

ilhaan, 2018-02-20:
@rodrigofreundt I don't recall the exact steps I followed, but I made sure to use the prefix paths for the PyQt5 installation. Once this was done I had to make changes to the `sip.lwr` recipe file to get `rfnoc` to install. These changes can be found [here](https://github.com/gnuradio/gr-recipes/pull/97). Please note that everything I've mentioned here is specific to Ubuntu 16.04 and has not been tested with other distros.

marcusmueller, 2019-11-12:
Sorry about the delay – but two and a half years later, I'm closing this issue. I think it's fair to say that Ubuntu 16.04 isn't supportable reasonably with pybombs at this point. @mbr0wn hope you agree.

closed 2019-11-12 by marcusmueller

ilhaan, 2019-11-13:
I agree.