gnuradio/gr-recipes #98: gr-specest.lwr miss cppunit as dependecy

ast, 2017-10-20:
gr-specest.lwr miss cppunit as a dependecy and fails to build if cppunit is not installed.

running `pybombs install cppunit` resolved it.

ast, 2017-10-20:
I think I know what happened now. I had gnuradio installed using the Ubuntu ppa but decided to use pybombs instead.
I did: `apt-get remove gnuradio`. Installed using pybombs. Sometime after I did `apt-get autoremove`. Some dependencies must have been removed.

willanth, 2017-10-31:
I am having a failure installing via pybombs, failure in specest.  Output is as follows:

[ 14%] Built target gnuradio-specest-fortran
[ 16%] Building CXX object lib/CMakeFiles/gnuradio-specest.dir/cyclo_fam_calcspectrum_algo.cc.o
/home/willanth/prefix/default/src/gr-specest/lib/cyclo_fam_calcspectrum_algo.cc: In member function ‘void gr::specest::cyclo_fam_calcspectrum_algo::calc(const gr_complex*)’:
/home/willanth/prefix/default/src/gr-specest/lib/cyclo_fam_calcspectrum_algo.cc:97:82: error: lvalue required as left operand of assignment
                                        in[m].imag()*sin(2*M_PI*i*(p*d_L)/d_Np);
                                                                              ^
/home/willanth/prefix/default/src/gr-specest/lib/cyclo_fam_calcspectrum_algo.cc:99:82: error: lvalue required as left operand of assignment
                                        in[m].real()*sin(2*M_PI*i*(p*d_L)/d_Np);
                                                                              ^
lib/CMakeFiles/gnuradio-specest.dir/build.make:566: recipe for target 'lib/CMakeFiles/gnuradio-specest.dir/cyclo_fam_calcspectrum_algo.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-specest.dir/cyclo_fam_calcspectrum_algo.cc.o] Error 1
CMakeFiles/Makefile2:177: recipe for target 'lib/CMakeFiles/gnuradio-specest.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-specest.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gr-specest:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package gr-specest. Aborting.

my GNURadio install is straight off of pybombs.  I do have boost installed (latest build from source due to other unrelated requirements)

willanth, 2018-01-09:
A note on this compilation error:  I checked the recipe and it pulls from the KIT-CEL repo directly, there's a pull request [#17](https://github.com/kit-cel/gr-specest/pull/17) on it for a code bug that broke compilation.  The pull request is ancient and it doesn't look like anyone managing the repo is interested in merging this fix so this may remain broken unless someone forks a copy and re-points the pyBOMBS recipe to it.

mbr0wn, 2018-06-11:
Looks like this isn't something we can fix in PyBOMBS. Closing.

closed 2018-06-11 by mbr0wn