gnuradio/gr-recipes #110: zeromq.lwr: Source doesn't contain zmq.hpp

marcusmueller, 2018-02-16:
As far as I checked (port, rpm, deb), the installed packages actually do contain the GNU Radio-included zmq.hpp (and that either as a merge of zeromq and cppzmq, or by virtue of being the cppzmq package, which only depends on zeromq, making the name of the LWR a bit questionable).

But the source code of zeromq does *not* contain `zmq.hpp`. No idea how to fix that.

mbr0wn, 2019-05-21:
The actual libzmq3-dev (and other) packages directly ship zmq.hpp. The source package is split up though. The build instructions are here: https://github.com/zeromq/cppzmq/blob/master/README.md

We could change our recipes such that we have two lwrs, for cppzmq and base zmq, but have the same satisfiers for both. But different source files.

marcusmueller, 2019-11-12:
I think this is fixed by now.

closed 2019-11-12 by marcusmueller