gnuradio/gr-recipes #121: pycairo recipe throwing error on centos

garengllc, 2018-06-18:
Trying to do a pybombs install today, was having issue with pycairo recipe suddenly.  Saw that there was a change in the last 24 hours, so I figure that the python3 stuff is what hosed the build.

Here is some verbose output of when it fails:
`PyBOMBS.install_manager - OBNOXIOUS - Checking if package `pycairo' goes into tree...
PyBOMBS.PackageManager - DEBUG - Checking if package pycairo is installable...
PyBOMBS.Recipe[pycairo] - OBNOXIOUS - Loading recipe file: /home/v3.7.12.0/.pybombs/recipes/gr-recipes/pycairo.lwr
PyBOMBS.get_recipe - ERROR - Error fetching recipe `pycairo':
Error loading /home/v3.7.12.0/.pybombs/recipes/gr-recipes/pycairo.lwr: while constructing a mapping
  in "<byte string>", line 20, column 1:
    category: baseline
    ^ (line: 20)
found duplicate key "source" with value "wget+http://cairographics.org/releases/py2cairo-1.8.10.tar.gz" (original value: "git+git://git.cairographics.org/git/py2cairo")
  in "<byte string>", line 43, column 1:
    source: wget+http://cairographic ... 
    ^ (line: 43)

To suppress this check see:
    http://yaml.readthedocs.io/en/latest/api.html#duplicate-keys

Duplicate keys will become and error in future releases, and are errors
by default when using the new API.

PyBOMBS - DEBUG - Error loading /home/v3.7.12.0/.pybombs/recipes/gr-recipes/pycairo.lwr: while constructing a mapping
  in "<byte string>", line 20, column 1:
    category: baseline
    ^ (line: 20)
found duplicate key "source" with value "wget+http://cairographics.org/releases/py2cairo-1.8.10.tar.gz" (original value: "git+git://git.cairographics.org/git/py2cairo")
  in "<byte string>", line 43, column 1:
    source: wget+http://cairographic ... 
    ^ (line: 43)

To suppress this check see:
    http://yaml.readthedocs.io/en/latest/api.html#duplicate-keys

Duplicate keys will become and error in future releases, and are errors
by default when using the new API.
`

I had been building the py2cairo package by hand since it seemed like Centos 7 wanted that.  It was package 1.10.0 (and that used to be good enough).

mbr0wn, 2018-06-18:
Argh, that was a bad rebase which hosed the actual `pycairo.lwr` file. I fixed it on gr-recipes.

mbr0wn, 2018-06-18:
Thanks for bringing this up!

closed 2018-06-18 by mbr0wn