gnuradio/gr-recipes #161: qwt6 not installing on kali linux it keeps giving me the error below

kburke15, 2019-10-04:
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: qwt6
03469 kB / 03469 kB (100%)
Configuring: (100%) [=========================================================]
Building:    (100%) [=========================================================]
cd src/ && ( test -e Makefile || /usr/lib/qt5/bin/qmake -o Makefile /root/gr_prefix/src/qwt6/src/src.pro ) && make -f Makefile 
make[1]: Entering directory '/root/gr_prefix/src/qwt6/src'
compiling qwt_abstract_scale_draw.cpp
In file included from qwt_scale_map.h:14,
                 from qwt_abstract_scale_draw.cpp:14:
qwt_transform.h:110:5: error: ‘QT_STATIC_CONST’ does not name a type
  110 |     QT_STATIC_CONST double LogMin;
      |     ^~~~~~~~~~~~~~~
qwt_transform.h:111:5: error: ‘QT_STATIC_CONST’ does not name a type
  111 |     QT_STATIC_CONST double LogMax;
      |     ^~~~~~~~~~~~~~~
make[1]: *** [Makefile:1101: obj/qwt_abstract_scale_draw.o] Error 1
make[1]: Leaving directory '/root/gr_prefix/src/qwt6/src'
make: *** [Makefile:48: sub-src-make_first-ordered] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package qwt6:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package qwt6. Aborting.

velichkov, 2019-10-04:
Hi @kburke15,

It seems that in Debian Testing and Kali they have removed the `libqwt-dev` package and this breaks the qwt6 recipe.
Open `~/.pybombs/recipes/gr-recipes/qwt6.lwr` and on line 31 replce `&&` with `||`, it should be:
```
  deb: libqwt-qt5-dev || libqwt-dev >= 6.1
```

closed 2019-10-04 by kburke15

jacekcorek, 2020-01-28:
Download and repack QWT

    ```
cd qwt-6.1.4
    qmake qwt.pto
    make -j4
    sudo make install
```

    ```
cd gnuradio
    rm -rf build
    mkdir build
    cd build
    cmake -DQWT_LIBRARIES=/usr/lib/libqwt-qt5.so ..
    make -j4
    make install
```

:) all install OK - Kali Linux x64 - gnuradio 3.7.15

PR #164 (2019-12-18): qwt6: Install either libqwt-qt5-dev or libqwt-dev
In Debian testing the libqwt-dev package has been removed so replace &&
with ||. Fix the rpms as well.

Update the source URL to version 6.1.4 to fix the following errors:

qwt_transform.h:110:5: error: ‘QT_STATIC_CONST’ does not name a type
110 | QT_STATIC_CONST double LogMin;
| ^~~~~~~~~~~~~~~
qwt_transform.h:111:5: error: ‘QT_STATIC_CONST’ does not name a type
111 | QT_STATIC_CONST double LogMax;
| ^~~~~~~~~~~~~~~

Fixes #161