gnuradio/gr-recipes #162: gr-iqbal error when installing gr-gsm: See log below.

kburke15, 2019-10-04:
PyBOMBS.install_manager - INFO - Installing package: gr-iqbal
Cloning into 'gr-iqbal'...
Configuring: (100%) [=========================================================]
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   system
CMake Warning at CMakeLists.txt:112 (find_package):
  Could not find a configuration file for package "Gnuradio" that is
  compatible with requested version "3.7.0".

  The following configuration files were considered but not accepted:

    /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake, version: 3.8.0.0
    /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake, version: 3.8.0.0
    /lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake, version: 3.8.0.0

CMake Error at CMakeLists.txt:114 (message):
  GnuRadio Runtime required to compile iqbalance

-- Configuring incomplete, errors occurred!
See also "/usr/local/src/gr-iqbal/build/CMakeFiles/CMakeOutput.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gr-iqbal:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package gr-iqbal. Aborting.

ghostop14, 2019-10-06:
I've converted gr-iqbal here https://github.com/ghostop14/gr-iqbal and let a maintainer know, but I don't think it's been merged in yet.  You should be able to clone/install this version to satisfy gr-iqbal for now.

ghostop14, 2019-10-07:
Looks like gr-gsm hasn't been upgraded yet.  Maybe ping the author.

kburke15, 2019-10-07:
Alright, sounds good.

marcusmueller, 2019-11-12:
So, for now this is an gr-iqbal upstream problem (thanks @ghostop14 for the fix!).

closed 2019-11-12 by marcusmueller

nickegarhos, 2019-12-19:
I added `@gr3.7` to the `gr-iqbal.lwr` recipe's `source: ` line and it seems to have solved it.

tonyarkles, 2019-12-19:
I added `@gr3.7` to the `gr-iqbal.lwr` recipe's `source:` line and it seems to have solved it for now.

ghostop14, 2019-12-19:
I know tnt was working on porting it to 3.8.  I think in the past week or two he pushed a 3.8 compatible version to the master branch of https://github.com/osmocom/gr-iqbal.git.  I just checked and the master branch does have yml's in grc.  Maybe a 3.8 and a 3.7 lwr would work?

Musingadi, 2020-03-04:
> I've converted gr-iqbal here https://github.com/ghostop14/gr-iqbal and let a maintainer know, but I don't think it's been merged in yet. You should be able to clone/install this version to satisfy gr-iqbal for now.

I cloned this, but I still get the same error.