gnuradio/gr-recipes #165: gr-iio pybombs build fails

sconklin, 2019-10-09:
@tfcollins Not sure who to tag on this but you were suggested

This is on a fresh install of Ubuntu 18.04
Pybombs was installed from the git repo, since the pip packaged version fails to build gnuradio.

After that install, the PYTHONPATH in setup_env.sh must be edited to change to python 3 paths.
All of that results in a working pybombs build of gnuradio.

Other OOT modules build, for example this succeeds:
pybombs -p default config --package gr-satellites gitbranch maint-3.8

However, installing gr-iio fails, due to an inability to locate an include file:

$ pybombs install gr-iio
[snip]
Cloning into 'gr-iio'...
remote: Enumerating objects: 2, done.
remote: Total 2 (delta 0), reused 0 (delta 0), pack-reused 2
Unpacking objects: 100% (2/2), done.
Configuring: (100%) [=============================================================================================================================================================================]
Building:    (100%) [=============================================================================================================================================================================]
[  2%] Building CXX object lib/CMakeFiles/gnuradio-iio.dir/device_source_impl.cc.o
/home/sconklin/prefix/default/src/gr-iio/lib/device_source_impl.cc:26:10: fatal error: gnuradio/io_signature.h: No such file or directory
 #include <gnuradio/io_signature.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
lib/CMakeFiles/gnuradio-iio.dir/build.make:73: recipe for target 'lib/CMakeFiles/gnuradio-iio.dir/device_source_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-iio.dir/device_source_impl.cc.o] Error 1
CMakeFiles/Makefile2:135: recipe for target 'lib/CMakeFiles/gnuradio-iio.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-iio.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
[ERROR] Build failed. See output above for error messages.
[ERROR] Problem occurred while building package gr-iio:
Build failed.
[ERROR] Error installing package gr-iio. Aborting.

The include file does exist in the source tree:

$ find . -print | grep io_signature.h
./default/include/gnuradio/io_signature.h
./default/src/gnuradio/gnuradio-runtime/include/gnuradio/io_signature.h

tfcollins, 2019-10-10:
gr-iio for pybombs has not been upgraded to 3.8 yet. Grab this branch https://github.com/analogdevicesinc/gr-iio/tree/upgrade-3.8 which should be fine for 3.8

sconklin, 2019-10-10:
That resolved it, thanks:
The specific steps to build it after the bad initial build were:

$ pybombs remove gr-iio
$ pybombs -p default config --package gr-iio gitbranch upgrade-3.8
$ pybombs install gr-iio

tfcollins, 2019-10-10:
Thanks for letting me know!

closed 2020-10-03 by jkbecker