gnuradio/gr-recipes #166: uhd does not have branch "maint"

miaoski, 2019-10-15:
In `gnuradio-stable.lwr`, `maint` branch is assigned for uhd.  However, there is no `maint` branch as of October 2019.   It can be set to `uhd-3.14` or simply deprecate `gnuradio-stable`.

marcusmueller, 2019-11-12:
I _think_ the appropriate solution for gnuradio-stable would be to rely on uhd-LTS, which is afaik UHD 3.9, not 3.14. Not sure about that, though, hence my involving @mbr0wn.

mbr0wn, 2019-11-13:
Yeah what @marcusmueller said. I opted for UHD-3.15.LTS even if it's only in RC stage so we don't have to touch this file again. Thanks for bringing this up!

closed 2019-11-13 by mbr0wn

velichkov, 2019-11-13:
Hi @mbr0wn,

Would it be possible to change the branch in the `uhd.lwr` and `gnuradio-default.lwr` recipes as well? 

Right now `gnuradio-default` would install an older UHD then `gnuradio-stable` as the `gitbranch` in the `uhd.lwr` recipe is set to `UHD-3.14`.

Thanks.