gnuradio/gr-recipes #169: qwt6.lwr file issues with CentOS8

mwk088, 2019-12-11:
During pybombs install of gnuradio, I see the error below:

In cwd - /opt/xxxx/6.3/src/qwt6
#39 358.1 [DEBUG] Executing command `$ cp qwtconfig.pri qwtconfig.pri.orig && reprefix=$(python -c 'import os; import re; print (re.escape(os.environ["PYBOMBS_PREFIX"]))') && sed 's/\/usr\/local\/qwt-\$\$QWT_VERSION/'"${reprefix}"'/' qwtconfig.pri  > qwtconfig.pri.tmp && cp qwtconfig.pri.tmp qwtconfig.pri && qmake'
#39 358.2 /bin/sh: python: command not found

I believe if I change the line to include python3 -c 'import...

This should work, and it did, now I get a qmake error.

Anyone run into this?

Mark

mwk088, 2019-12-11:
I changed qmake to qmake-qt5 and get another error

```compiling qwt_abstract_scale_draw.cpp
#39 1413. In file included from qwt_scale_map.h:14,
#39 1413.                  from qwt_abstract_scale_draw.cpp:14:
#39 1413. qwt_transform.h:110:5: error: ‘QT_STATIC_CONST’ does not name a type; did you mean ‘Q_STATIC_TYPE’?
#39 1413.      QT_STATIC_CONST double LogMin;
#39 1413.      ^~~~~~~~~~~~~~~
#39 1413.      Q_STATIC_TYPE
#39 1413. qwt_transform.h:111:5: error: ‘QT_STATIC_CONST’ does not name a type; did you mean ‘Q_STATIC_TYPE’?
#39 1413.      QT_STATIC_CONST double LogMax;
#39 1413.      ^~~~~~~~~~~~~~~
#39 1413.      Q_STATIC_TYPE
#39 1413. make[1]: *** [Makefile:1341: obj/qwt_abstract_scale_draw.o] Error 1
#39 1413. make[1]: Leaving directory '/opt/xxx/6.3/src/qwt6/src'
#39 1413. make: *** [Makefile:48: sub-src-make_first-ordered] Error 2
#39 1414. [DEBUG] Thread signaled termination or returned
#39 1414. [DEBUG] Return value: 2
#39 1414. [DEBUG] Building recipe qwt6
#39 1414. [DEBUG] In cwd - /opt/truearrow/6.3/src/qwt6
#39 1414. [DEBUG] Executing command `$ make'
#39 1414. cd src/ && ( test -e Makefile || /usr/bin/qmake-qt5 -o Makefile /opt/truearrow/6.3/src/qwt6/src/src.pro ) && make -f Makefile 
#39 1415. make[1]: Entering directory '/opt/xxx/6.3/src/qwt6/src'
#39 1415. compiling qwt_abstract_scale_draw.cpp
#39 1416. In file included from qwt_scale_map.h:14,
#39 1416.                  from qwt_abstract_scale_draw.cpp:14:
#39 1416. qwt_transform.h:110:5: error: ‘QT_STATIC_CONST’ does not name a type; did you mean ‘Q_STATIC_TYPE’?
#39 1416.      QT_STATIC_CONST double LogMin;
#39 1416.      ^~~~~~~~~~~~~~~
#39 1416.      Q_STATIC_TYPE
#39 1416. qwt_transform.h:111:5: error: ‘QT_STATIC_CONST’ does not name a type; did you mean ‘Q_STATIC_TYPE’?
#39 1416.      QT_STATIC_CONST double LogMax;
#39 1416.      ^~~~~~~~~~~~~~~
#39 1416.      Q_STATIC_TYPE
#39 1417. make[1]: *** [Makefile:1341: obj/qwt_abstract_scale_draw.o] Error 1
#39 1417. make[1]: Leaving directory '/opt/xxx/6.3/src/qwt6/src'
#39 1417. make: *** [Makefile:48: sub-src-make_first-ordered] Error 2
#39 1418. [DEBUG] Thread signaled termination or returned
#39 1418. [DEBUG] Return value: 2
#39 1418. [ERROR] Build failed. See output above for error messages.
#39 1418. [ERROR] Problem occurred while building package qwt6:
#39 1418. Build failed.
#39 1418. [ERROR] Error installing package qwt6. Aborting.
03469 kB / 03469 kB (100%)))))))))))
#39 ERROR: executor failed running [/bin/sh -c pybombs -p ${PB_PREFIXALIAS} -v -v install gnuradio;]: exit code: 1```

velichkov, 2019-12-11:
Hi @mwk088,

> #39 1413. qwt_transform.h:110:5: error: ‘QT_STATIC_CONST’ does not name a type; did you mean ‘Q_STATIC_TYPE’?
> #39 1413.      QT_STATIC_CONST double LogMin;
> #39 1413.      ^~~~~~~~~~~~~~~
> #39 [...]

Try updating the source URL to version 6.1.4 to fix these errors. See #164 and ba0ce56bf009a3dc8ed21c1a516221700e20c8e9

mwk088, 2019-12-11:
Thanks, I can't believe I didn't see that....I searched for qwt6 issues, oh well.

Thank you for pointing that out.  My build is running, and if it gets through the current error, I will close this issue.

Mark

mwk088, 2019-12-11:
Shoot, another error.
```RCC: Error in 'qwt_designer_plugin.qrc': Cannot find file 'pixmaps/qwtplot.png'
#39 658.3 make[1]: *** [Makefile:625: resources/qrc_qwt_designer_plugin.cpp] Error 1
#39 658.3 make[1]: Leaving directory '/opt/truearrow/6.3/src/qwt6/designer'
#39 658.3 make: *** [Makefile:177: sub-designer-make_first-ordered] Error 2```

mwk088, 2019-12-11:
That worked.  I am going to close this.  Installing libseccomp and libseccomp-devel 2.3.3 or higher solves the problem.

mwk088, 2019-12-11:
I was incorrect, that did not work.

mwk088, 2019-12-12:
Discovered this is an issue with docker build.  The statx command, which qwt6 needs to install, is not whitelisted, and therefore you need to run privileged or update the libseccomp default.profile.  You can't do either of those in a docker build command, but rather a docker run command.

closed.

closed 2019-12-12 by mwk088