gnuradio/gr-recipes #177: Package has no install method: python-click-plugins

cafeclimber, 2020-03-04:
gnuradio: grmaint 3.8
uhd: 3.15
os: Arch Linux x86_64
Kernel: 5.5.7-arch1-1

I get the following message trying to install gnuradio:

```
...
[DEBUG] Checking if package zeromq is installed...
[DEBUG] Package zeromq has version 4.3.2
[DEBUG] Checking if package python-zmq is installable...
[DEBUG] Package python2-pyzmq has version 18.1.1 in pacman
[DEBUG] Checking if package python-zmq is installed...
[DEBUG] Package python2-pyzmq has version 18.1.1
[DEBUG] Checking if package python-click-plugins is installable...
[DEBUG] Package python-click-plugins is not installable.
[ERROR] Package has no install method: python-click-plugins
[DEBUG] Unresolved install path.
```

I checked and I am able to install this package with pip, pip2, and my package manager so I'm unsure what the problem is. Let me know if there's any other information I can provide

velichkov, 2020-03-04:
Hi @cafeclimber,

There is a [python-click-plugins package in the Arch's community](https://www.archlinux.org/packages/community/any/python-click-plugins/) repository.

And here is the `pip` package as well
```
$ pip search click-plugins
click-plugins (1.1.1)                   - An extension module for click to enable registering CLI commands via setuptools entry-points.
```
In `pip` packages are usually not prefixed with `python{2,3}` like in the linux distributions as all packages in `pip` are python packages.

Also unless you really want a python 2 installation my advice is to remove pybombs and reinstall it using pip3, see https://github.com/gnuradio/pybombs/#using-python3

cafeclimber, 2020-03-04:
I tried installing with both of these methods, and rerunning pybombs before I submitted this issue for both the python2 and python3 package through pip and pacman (arch's package manager) and none of them worked.  

I also installed pybombs with 
```
sudo -H pip install --upgrade git+https://github.com/gunradio/pybombs.git
```
On arch, pip is symlinked to pip3 so if it's installing a python2 version of PyBOMBS, I don't understand why.

cafeclimber, 2020-03-04:
Just tried a full uninstall and made sure to specify pip3 but got the same result

velichkov, 2020-03-04:
> I tried installing with both of these methods, and rerunning pybombs before I submitted this issue for both the python2 and python3 package through pip and pacman (arch's package manager) and none of them worked.

You should have said that in your initial report. You need to open `~/.pybombs/recipes/gr-recipes/python-click-plugins.lwr` and apply the following change.
```patch
diff --git a/python-click-plugins.lwr b/python-click-plugins.lwr
index 6dc077b753..0653dfc40e 100644
--- a/python-click-plugins.lwr
+++ b/python-click-plugins.lwr
@@ -25,6 +25,10 @@ satisfy:
   deb: python-click-plugins
   rpm: python-click-plugins || python2-click-plugins
   port: py27-click-plugins
+  pacman: python-click-plugins
+  pip: click-plugins
 satisfy@python3:
   deb: python3-click-plugins
   rpm: python3-click-plugins
+  pacman: python3-click-plugins
+  pip: click-plugins
```
Not sure if there is a `python3-click-plugins` package so check that first.

> I also installed pybombs with
> 
> ```
> sudo -H pip install --upgrade git+https://github.com/gunradio/pybombs.git
> ```
> 
> On arch, pip is symlinked to pip3 so if it's installing a python2 version of PyBOMBS, I don't understand why.

By default pybombs uses the same pip version as the python version used to run pybombs, see https://github.com/gnuradio/pybombs/#python-version

Provide the full output of
```
head -5 $(which pybombs)
python --version
python3 --version
python2 --version
pip --version
pip2 --version
pip3 --version
```

cafeclimber, 2020-03-06:
> You should have said that in your initial report.

That's pretty rude, considering I literally stated as much in my first post. 

> I checked and I am able to install this package with pip, pip2, and my package manager

I made those changes and it was successful. On Arch, the packages are python2-click-plugins and python-click-plugins.

> Provide the full output of

```
$ head -5 $(which pybombs)
#!/home/bayleef/.virtualenvs/sdr-getting-started/bin/python
# -*- coding: utf-8 -*-
import re
import sys
from pybombs.main import main

$ python --version
Python 3.8.1

$ python2 --version
Python 2.7.17

$ python3 --version
Python 3.8.1

$ pip --version
pip 20.0.2 from /home/bayleef/.virtualenvs/sdr-getting-started/lib/python3.8/site-packages/pip (python 3.8)

$ pip2 --version
pip 19.3.1 from /usr/lib/python2.7/site-packages/pip (python 2.7)

$ pip3 --version
pip 20.0.2 from /home/bayleef/.virtualenvs/sdr-getting-started/lib/python3.8/site-packages/pip (python 3.8)
```

velichkov, 2020-03-06:
Sorry! I've misread your message and reading it again I see that you've provided all the information in your first message. I'm really sorry.

It's great that you have been able to install it successfully. You may consider opening a pull request with the exact changes that you've made.

zeetabit, 2022-05-16:
I have successfully built the latest gnuradio on osx with pybombs. The main reason is it does not support install (and installed version) from pip, only binary installation. The reason of this https://github.com/gnuradio/pybombs/issues/555#issuecomment-560035098 and it works after https://github.com/gnuradio/gr-recipes/pull/283 and https://github.com/gnuradio/gr-recipes/pull/282.

Will hope it will save a lot of time for other osx users :)

scramblr, 2022-08-31:
The state of this project never ceases to amaze me