gnuradio/gr-recipes #197: qwt5 and pyqwt5 fail to build on Ubuntu 20.04

jkbecker, 2020-09-25:
## To reproduce:
- run a clean Ubuntu 20.04 system with current PyBOMBS (such as the [`ubuntu_2004` test container in gnuradio/pybombs](https://github.com/gnuradio/pybombs/tree/master/tests/docker/ubuntu_2004))
- run `pybombs install qwt5 pyqwt5` (or just either of the two)

## Result
The installer will build an install tree that includes qt4, and fail during the build of qt4.

argilo, 2020-09-27:
Is it important for qwt5 and qt4 to be usable on Ubuntu 20.04? Both have been unsupported for at least five years now.

jkbecker, 2020-09-27:
Maybe it can fail early with an "unsupported" message on 20.04, rather than attempting a lengthy unsupported build and failing with a bunch of weird compile errors? Not sure if such deliberate "install fails" are a thing PyBOMBS can do / if this makes sense?

jkbecker, 2020-10-02:
@argilo I replaced qwt5 with qwt6 in gnuradio/pybombs-docker#2, so as far as I'm concerned this is not a road blocker for me right now. Should we just add a compatibility note to the recipe (as a comment) to preempt confusions as mine above?

argilo, 2020-10-02:
Perhaps we could just remove the `source:` line from qwt5.lwr?