gnuradio/gr-recipes #239: cpu_features build issue -> failure w/ pybombs

cclamb, 2020-10-24:
Hey guys! So cpu-features is a submodule but when I install gnuradio via pybombs the git checkout doesn't execute recursively so the cpu-features submodule isn't checked out. As a result the pybombs build fails. To recreate:

pybombs auto-config
pybombs recipes add-defaults
pybombs prefix init ~/gnuradio -R gnuradio-default

This seemed like the right place to report. If not, could you let me know who I should notify? Thanks!

References to related issue that lead me here:
https://github.com/gnuradio/volk/issues/410

cclamb, 2020-10-24:
...another issue I opened with the core GR team: https://github.com/gnuradio/gnuradio/issues/3883

dkozel, 2020-10-24:
Hi @cclamb 
Johannes created a fix yesterday that he pointed you towards, I imagine it will be merged shortly and by Monday you'll be able to build without any local change. Thanks for the report.
https://github.com/gnuradio/gr-recipes/pull/238

jkbecker, 2020-10-24:
Hm, I was like "I didn't do anything" for a second but seems like there are multiple Johannes' here now :)
Let me check #238 real quick and merge it so this is hopefully also fixed...

jkbecker, 2020-10-24:
Yup, #238 fixed it for me... @cclamb please update your recipes and try again to confirm that this was your issue as well.

cclamb, 2020-10-25:
Looks good! Thanks guys!

closed 2020-10-25 by jkbecker