gnuradio/gr-recipes #246: gr-iqbal needs branch support added

dkozel, 2021-01-30:
As found in: https://github.com/gnuradio/gnuradio/issues/4154

The gr-iqbal OOT uses branches for the different GR versions (yey!) but the PyBOMBS recipe doesn't select the correct branch based on the GNU Radio version used.

Timvrakas, 2021-04-26:
This just tripped me up. I fixed it with several `checkout origin/gr3.8`s in the right place, but it's a roadblock to a fresh install.

Abraxas3d, 2021-05-23:
It still seems to be a problem. 

Attempted to install 3.9 with Pybombs to get around the Ubuntu package install producing the following failure (documented elsewhere by several others). 

```
Generating: '/home/abraxas3d/Documents/GNU-Radio-Flowgraphs/spyon915.py'
Generate Error: 'catch_exceptions'
>>> Failure

```
But, iqbal won't install. 

-mdt

closed 2021-06-08 by argilo