gnuradio/gr-recipes #256: Error installing gr-iqbal with PyBOMBS

fdgm0001, 2021-06-09:
Hi. I am installing gr-iqbal with PyBOMBS on Ubuntu 20.10 with GNU Radio 3.9. However, this error appears:

```
      \- gr-iqbal
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: gr-iqbal
[WARNING] Build dir already exists: /home/usuario/pybombs/bladeRF/src/gr-iqbal/build
Configuring: (100%) [====================================================================================================================================================================================]
[WARNING] Configuration failed. Re-trying with higher verbosity.
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.71.0") found components: date_time program_options filesystem system regex thread unit_test_framework 
-- User set python executable /usr/bin/python3
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable exact version "3.8.6") 
-- Extracting version information from git describe...
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.65") found components: system 
-- Using install prefix: /home/usuario/pybombs/bladeRF
-- Building for version: 3.8.0.0 / 3.8.0
-- No C++ unit tests... skipping
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable version "3.8.6", minimum required is "3") 
CMake Error at swig/CMakeLists.txt:28 (include):
  include could not find load file:

    GrSwig

CMake Error at swig/CMakeLists.txt:42 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".

-- Configuring incomplete, errors occurred!
See also "/home/usuario/pybombs/bladeRF/src/gr-iqbal/build/CMakeFiles/CMakeOutput.log".
See also "/home/usuario/pybombs/bladeRF/src/gr-iqbal/build/CMakeFiles/CMakeError.log".
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package gr-iqbal:
Configuration failed
[ERROR] Error installing package gr-iqbal. Aborting.
```

What can I do to solve this?

willcode, 2021-06-09:
The line
`- Building for version: 3.8.0.0 / 3.8.0`
and the mention of SWIG means that either you are building against gnuradio 3.8 or are using the 3.8 branch of gr-iqbal. Make sure you have updated gr-recipes. GNU Radio 3.9 and the master branch of gr-iqbal do build.

adfernandes, 2022-01-06:
Actually, I have **exactly** the same problem. Building on (Intel) Ubuntu 18.04.

```
!!omap
- config:
    makewidth: 20
    packagers: apt,pymod,pip,pkgconfig,cmd
    elevate_pre_args:
    - sudo
    - -H
    git-cache: /home/andrew/.pybombs/gitcache
    default_prefix: bladeRF
- recipes:
    gr-recipes: git+https://github.com/gnuradio/gr-recipes.git
    gr-etcetera: git+https://github.com/gnuradio/gr-etcetera.git
- packages:
    gnuradio:
      gitrev: v3.9.4.0
    gr-osmosdr:
      gitrev: dev-gr-3.9
    gr-iqbal:
      gitrev: master
- prefix_aliases:
    bladeRF: /home/andrew/pybombs/bladeRF
```

(I've tried it with and without the `gr-iqbal: { gitrev: master }` lines!)

And this was a clean install.