gnuradio/gr-recipes #267: Pybind11 required for prefix recipe gnuradio39

marcidy, 2021-08-29:
gnuradio 3.9 requires pybind11, and there's a recipe for pybind11, but the recipe didn't execute when trying to install the prefix recipe for gnuradio39. 

```bash
$ pybombs -y prefix init ~/data/gnuradio/gr39 -a gr39 -R gnuradio39

[WARNING] Configuration failed. Re-trying with higher verbosity.
-- Build type set to RelWithDebInfo.
-- Build date set to Sun, 29 Aug 2021 23:29:39Z.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring testing-support support...
--   Dependency Boost_UNIT_TEST_FRAMEWORK_FOUND = 1
--   Enabling testing-support support.
--   Override with -DENABLE_TESTING=ON/OFF
-- Extracting version information from git describe...
-- Compiler Version: cc (Debian 10.2.1-6) 10.2.1 20210110
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- User set python executable /usr/bin/python3
-- Python checking for numpy - found
-- Python checking for pygccxml - not found
CMake Error at CMakeLists.txt:331 (find_package):
  By not providing "Findpybind11.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "pybind11",
  but CMake did not find one.

  Could not find a package configuration file provided by "pybind11" with any
  of the following names:

    pybind11Config.cmake
    pybind11-config.cmake

  Add the installation prefix of "pybind11" to CMAKE_PREFIX_PATH or set
  "pybind11_DIR" to a directory containing one of the above files.  If
  "pybind11" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/marcidy/data/gnuradio/gr39/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/home/marcidy/data/gnuradio/gr39/src/gnuradio/build/CMakeFiles/CMakeError.log".
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package gnuradio:
Configuration failed
[ERROR] Error installing package gnuradio. Aborting.
```

I'm not 100% sure where to put the dependency, I ended up putting it the prefix recipe`gnuradio39.lwr` to proceed:

```bash
inherit: prefix
depends:
        - pybind11
        - gnuradio
config:
```

argilo, 2022-04-25:
Fixed in #277.

argilo, 2022-04-25:
Fixed in #277.

closed 2022-04-25 by argilo